NebraLtd/Helium-Guides #1: Play with cache settings?

shawaj, 2021-01-18:
For example...
```
requestHeaders: {
    'cache-control': 'no-cache'
  }
```

shawaj, 2021-01-20:
Think we need to do any more stuff with caching etc? Or shall we close this @ryanteck ?

ryanteck, 2021-01-20:
I think it's all fine, the images are set to be cached on the server. Only the content isn't but it's flatfile so should be fine.

When we start shipping I can keep an eye and always tweak it to cache slightly more when the docs are more finished.

shawaj, 2021-01-20:
@ryanteck I think this setting actually relates to browser caching not server side

ryanteck, 2021-01-20:
I think it should be fine and will be best for Nginx to handle the caching, it doesn't cache on the server side but controls and tells the browser what to cache and how long for.

Roughly in a 60 second DOS test the Pi handled I think it was over 200,000 requests with only 2-3 of them failing.

shawaj, 2021-01-20:
Cool closing this then

closed 2021-01-20 by shawaj