NebraLtd/Helium-Guides #6: Coverpage?

ryanteck, 2021-01-18:
Would you like me to re-enable coverpage?

Here's what it would look like.
[image]

ryanteck, 2021-01-18:
@shawaj

shawaj, 2021-01-18:
@ryanteck what does it do exactly?

I was just thinking we populate the README.md for the home page as I think it defaults to that no?

ryanteck, 2021-01-18:
Just adds a different look above the readme as like an entry page.

ryanteck, 2021-01-18:
Pushed it so you can see what it looks like, can disable it by removing the line if don't like.

shawaj, 2021-01-18:
Ah right so it's like a customisable header for the homepage? But the readme still populates below?

ryanteck, 2021-01-18:
Yeah

shawaj, 2021-01-18:
@ryanteck hmm I quite like that.

Maybe needs some input from @JohnWhiting on that as he's the marketing guru 😜

JohnWhiting, 2021-01-19:
@ryanteck @shawaj This looks great!

closed 2021-01-19 by shawaj