NebraLtd/Helium-Guides #7: PDF downloads

shawaj, 2021-01-18:
See plugin here https://github.com/meff34/docsify-to-pdf-converter

Also docsify awesome https://github.com/docsifyjs/awesome-docsify

ryanteck, 2021-01-18:
Adding it now

ryanteck, 2021-01-18:
https://github.com/NebraLtd/Helium-Guides/blob/main/pdf/readme.pdf

shawaj, 2021-01-19:
@ryanteck how do we generate them for other pages? And does this auto-generate somehow or we can build into a GitHub workflow?

shawaj, 2021-01-19:
From here - https://github.com/GoogleChrome/puppeteer/blob/master/docs/api.md#pagepdfoptions

And here - https://github.com/GoogleChrome/puppeteer/blob/master/docs/api.md#pageemulatemediamediatype

It looks like there is quite a lot of customisation options, including separate print templates etc. 

Wonder if we can have pdfs of just certain pages / folders from the repo

ryanteck, 2021-01-19:
> @ryanteck how do we generate them for other pages? And does this auto-generate somehow or we can build into a GitHub workflow?

Currently for testing I've been running a command locally.

I can likely set it up on github actions or a cron that runs on the Pi that hosts it.

ryanteck, 2021-01-20:
PDF generation files because of removing the # from what I can tell.

https://github.com/meff34/docsify-to-pdf-converter/blob/aae1db48cce15ebfae744bb0a046cf5c68f56f1e/src/render.js#L22

shawaj, 2021-01-20:
Nice find @ryanteck 

Started an issue https://github.com/meff34/docsify-to-pdf-converter/issues/24

Maybe we can do a pr that checks for the history router and changes the behaviour?

ryanteck, 2021-01-20:
It looks like an abandoned project now so don't think it'll get fixed.

I think the easiest thing will be if I just edit the # out and see if that fixes it.

shawaj, 2021-01-20:
Yeah for sure. So just fork it and do our own?

ryanteck, 2021-01-20:
I don't think should need to do that.

The plugin is installed locally on the server so I just need to remove the # from that. 

I'll then set it up to automate generation of the PDF each git push too

Just doing it now.

shawaj, 2021-01-20:
@ryanteck can we just do one PDF from the whole site or we can do multiple different PDFs?

ryanteck, 2021-01-20:
I think with this plug in it is the whole site.

It might be better instead to use a different piece of software to render induvidual PDFs from each markdown. 

I'm currently having issues getting the plugin set up on the Pi server because of lack of IPv4 is causing issues with NPM.

shawaj, 2021-01-20:
Yeah, I guess this will do for now and we can improve in the future

ryanteck, 2021-04-25:
Was still having issues with this plugin.

For now I've configured the server that it will generate PDFs for files listed in https://github.com/NebraLtd/Helium-Guides/blob/main/pdf-gen/gen-pdfs.sh

Currently I've just got it generating the main ones that need to be provided as PDF which are the overview and quick start guides.

Could possibly in the future move it to be where it generates for every file, however I think for most just the main guides are the ones required to be in PDF form.

These are generated every time the repo's main branch is pushed to, and are uploaded to https://helium.nebra.com/pdfs/

closed 2021-04-25 by ryanteck