NebraLtd/Helium-Guides #14: Add Facebook messenger chat link

shawaj, 2021-01-19:
Similar to what is on nebra.com

Image:
[image]

Links to https://m.me/1469008860077745?ref=messenger_commerce_1163199097047119_https://www.nebra.com/collections/helium-hotspot-miners-hnt

shawaj, 2021-01-19:
Also possibly add contact form etc

ryanteck, 2021-01-19:
Hmm, the only thing with this could be rather than people reading the guides I can see them being lazy and messaging asking questions which would be in the docs?

We did talk about the support link being https://nebra.io/helium-support , could be better to put it on that maybe?

Not sure if I'm thinking wrong though so ignore me if I am.

shawaj, 2021-01-19:
@ryanteck yeah good point actually, I hadn't thought of that. I think you are right. Lets have a support page that pushes people to read the docs etc first and use the search functionality.

@JohnWhiting I have added a short URL https://nebra.io/helium-support that points to https://helium.nebra.com/support - the page doesn't link to anything just yet as we haven't populated the page but will do

commit 5c162a9e (2021-02-01): Add updated info

Closes: #14
closed 2021-02-01 by shawaj