NebraLtd/Helium-Guides #17: Helium Distributors

JohnWhiting, 2021-01-19:
@shawaj Reckon we can add a section for distributors?

ryanteck, 2021-01-19:
I've added this as a rough template to begin with to see what it looks like.

shawaj, 2021-01-19:
No reason not to @JohnWhiting

Could always add it as a table or something

shawaj, 2021-01-19:
If you want to, for now you could just create a list on that page @JohnWhiting and we can work on making it pretty later...

shawaj, 2021-01-19:
Link https://github.com/NebraLtd/Helium-Guides/blob/main/docs/distributors.md

shawaj, 2021-01-19:
And then you can get your face on here @JohnWhiting ...

[image]

😜😜😜

ryanteck, 2021-01-19:
Would this be the better page for the buy now button to link to?

shawaj, 2021-01-19:
That is a good question @ryanteck - see what @JohnWhiting thinks I guess?

shawaj, 2021-01-31:
Fixed via https://github.com/NebraLtd/Helium-Guides/commit/e53da31c4637909446017274a5d6a228e1faad10#diff-e5c01616d31cbeb7a4c977fe83b5e5f3658f54651c479586523dba89dd5c8858

@JohnWhiting are we missing anyone here?

shawaj, 2021-01-31:
Closing in favour of #21

closed 2021-01-31 by shawaj