NebraLtd/Helium-Guides #20: Metadata issue?

shawaj, 2021-01-31:
When adding links to these docs to other websites, it seems to pull in the main meta description and title not the one of the page.

Maybe a settings issue?

ryanteck, 2021-01-31:
I think that could be a limitation of docsify as it's currently setup.

From what I can tell the details are generates and filled in when the page loads using JS. However for meta information on embeds it has to be done server side I believe.

There is a method by the looks of it to do server side rendering with docsify which I can setup. However it would mean the server has to do slightly more work per request but I think it shouldn't be much of an issue.

Shall I give server side rendering a try?

ryanteck, 2021-01-31:
IPv6 hassle is causing issues with docsify downloading packages it needs to run serverside. To fix this I'd need it moving to a proper VPS rather than the Pi we're using currently.

shawaj, 2021-01-31:
Maybe not really such a big deal. I guess we could just put it on our VPS although that's running Apache bit nginx so might be a pain.

What's the issue with the package downloads?

ryanteck, 2021-01-31:
NPM doesn't like IPv6 so I need a proper server with IPv4 and not a Pi.

shawaj, 2021-02-01:
@ryanteck ah ok...so even with the reverse proxy it's getting angry?

shawaj, 2021-02-01:
@ryanteck maybe this will help? https://npm.community/t/npm-fails-to-install-on-ipv6-only-networks/9508/3

shawaj, 2021-02-01:
This seems to be the core issue https://github.com/nodejs/node/pull/31567

ryanteck, 2021-02-01:
I tried a few of the fixes but still had issues, the core issue is only renting a server with IPv6 as I've never had an issue with any other hosting service.

tvoverbeek, 2021-02-01:
Do not know it this helps, but here a way to disable IPv6 on Debian (like Raspberry Pi):
https://www.itzgeek.com/how-tos/linux/debian/how-to-disable-ipv6-on-debian-9-ubuntu-16-04.html

ryanteck, 2021-02-01:
@tvoverbeek the issue is that the Pi rented out to run the site only has IPv6 connectivity. Disabiling it would brick it.

NPM is slightly buggy with this IPv6 only config as if it gets an IPv4 address returned from DNS it defaults to IPv4.

I can spend time working around it, but at the same time a Pi isn't really suitable, is slower and overall worse than renting a proper VPS (which gives backups etc).

ryanteck, 2021-02-01:
Looks like there's some tweaks I need to make to get SSR working even locally. 

```
ryan@ryan-B450-AORUS-M:~/Helium-Guides$ npm start

> helium-guides@ start /home/ryan/Helium-Guides
> docsify start . -c ssr.config.js

[SSR] Serving . now.
Listening at http://localhost:4000

TypeError: Cannot read property 'indexOf' of undefined
    at AbstractHistory.parse (/home/ryan/Helium-Guides/node_modules/docsify-server-renderer/build.js:267:27)
    at Compiler.compile (/home/ryan/Helium-Guides/node_modules/docsify-server-renderer/build.js:813:37)
    at Renderer.<anonymous> (/home/ryan/Helium-Guides/node_modules/docsify-server-renderer/build.js:1384:28)
    at Generator.next (<anonymous>)

```

shawaj, 2021-02-01:
@ryanteck don't think it's a particular urgent issue to sort this metadata issue right now. Can wait.

Could move the docs to our VPS though if it's causing issues. No biggie

ryanteck, 2021-03-23:
Closing for now, unfortunately a limitation with docsify.

closed 2021-03-23 by ryanteck