NebraLtd/Helium-Guides #23: Website doesn't update after merging a PR?

shawaj, 2021-01-31:
@ryanteck for some reason after updating the distributors page via a PR from @JohnWhiting the website hasn't updated.

Any idea why?

shawaj, 2021-01-31:
Or maybe it was just the cache on my browser and I'm a total muppet...

ryanteck, 2021-01-31:
Yeah it looked like it triggered fine even with a PR request.

[image]

All I can think is that it was caching. Would you like me to for now disable caching on the guides while we're doing lots of changes?

shawaj, 2021-01-31:
@ryanteck nah it's probably ok just to leave it, especially as we aren't shipping for a while anyway.

ryanteck, 2021-01-31:
Ok :+1:

closed 2021-01-31 by ryanteck