NebraLtd/Helium-Guides #26: Add info about updates / remote management

shawaj, 2021-01-31:
Once we have this nailed down, add this info to the FAQ

salmanfarisvp, 2021-06-05:
@shawaj Is that about balena and https://dashboard.nebra.com/ ?

shawaj, 2021-06-05:
#67 was more for the dashboard really

This one was more intended to be a description of how the OTA updates work etc. 

Not super urgent in any case as it's more for the nerdy folks who like to dig deeper into the details etc 😊

salmanfarisvp, 2021-06-05:
Sure 👍, we can add this, please add a list of things need to add 

- [ ] How OTA Updates  Works

salmanfarisvp, 2021-09-20:
Added
- [X]  https://helium.nebra.com/handy-guides/local-diagnostics 
- [X] https://helium.nebra.com/handy-guides/dashboard
- [X] https://helium.nebra.com/security

closed 2021-10-04 by salmanfarisvp