NebraLtd/Helium-Guides #28: Add operating temperatures

suprnrdy, 2021-02-03:
Add operating temperatures for both indoor and outdoor units. This will help people understand the breaking point, especially when they want to put these in environments with extreme temperatures.

shawaj, 2021-02-03:
From cm3+ datasheet https://www.raspberrypi.org/documentation/hardware/computemodule/datasheets/rpi_DATA_CM3plus_1p0.pdf
[image]

For the other bits it will need @ryanteck

shawaj, 2021-02-03:
Lora module  -40~+85°C(Industrial Grade)
[LoRa_GL5712_Product Brief.pdf](https://github.com/NebraLtd/Helium-Guides/files/5917884/LoRa_GL5712_Product.Brief.pdf)

ryanteck, 2021-02-03:
Off the top of my head the RPi one is the lowest @shawaj so I think if we go for that for now that should be a good rule of thumb.

shawaj, 2021-02-03:
@ryanteck that's what I thought as I remember we went mostly for industrial -40 to 85

ryanteck, 2021-02-03:
Yeah, I think saying -20 to +70 would be a good range to say due to it being sealed inside the case.

suprnrdy, 2021-02-03:
So none of the networking components have lower ratings?

ryanteck, 2021-02-03:
The Lora side is -40 to +85,

Just checked the Ethernet circuitry and that's the same too.

I haven't checked every single component but I think it was the compute module had the smallest range. So decreasing it that bit more should cover it fine.

Still need to look into it slightly more.

shawaj, 2021-02-03:
USB / Ethernet HUB chip for reference https://www.mouser.co.uk/datasheet/2/146/xr22800-1661675.pdf
[image]

shawaj, 2021-02-03:
Crystal is -40 to 85 also
Inductor -25 to 85
buck converter -40 to 125
Schottky -55 to +125
Eth connector -40 to 85

So yeah basically -20 to 70 is fine to suggest IMO.

shawaj, 2021-03-11:
Indoor currently says 20 to 30 degrees C - which is the recommended ambient temperature range but it is unclear (also I think we should say 10 to 30 degrees C recommended.

We should also add a "absolute maximum / minimum operating temp" based on the chipsets etc

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/overview.md
https://helium.nebra.com/indoor-hotspot/overview

salmanfarisvp, 2021-04-29:
> We should also add a "absolute maximum / minimum operating temp" based on the chipsets etc

Which chipset did we need to look at? Compute module or Lora concentrator?

salmanfarisvp, 2021-09-19:
Added Rated Ambient Temperature as  -25-80oC - https://github.com/NebraLtd/Helium-Guides/commit/8d6fa872270939891fcac11e997c79bd1df75d3b

closed 2021-10-04 by salmanfarisvp