NebraLtd/Helium-Guides #30: Hotspot Hardware Setup - Add note to never power on device while antenna is off

suprnrdy, 2021-02-03:
https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md#preparing-your-nebra-indoor-hotspot

Add a note to never power the unit on without the antenna.

shawaj, 2021-02-03:
@suprnrdy does this cause some issue with Helium if you do? Or you just mean because it won't work properly?

ryanteck, 2021-02-03:
@shawaj It's actually to protect the LoRa module as it can cause damage if ran without an antenna connected.

ryanteck, 2021-02-03:
Fixed for indoor, leaving open to remember to put it in the outdoor's guide.

vpetersson, 2021-02-03:
> @shawaj It's actually to protect the LoRa module as it can cause damage if ran without an antenna connected.

Would it be possible to add a check for this in software? It does sound like an easy mistake to make when assembling it (e.g. not screwed in properly).

ryanteck, 2021-02-03:
Not that I know of @vpetersson.

vpetersson, 2021-02-03:
Yeah I thought that was a long shot, but just thought I'd ask. Any chance we could use a proxy for it, such as RSSI or similar?

ryanteck, 2021-02-03:
I think the issue is any software route would mean the module turns on and then essentially is at the stage where it could cause damage.

I also don't think we could use RSSI / Signal strength and run a scan on boot as the hotspot is the gateway so it wouldn't have anything to scan in comparison to wi-fi.

vpetersson, 2021-02-03:
Fair enough, @ryanteck.

shawaj, 2021-02-03:
Maybe there's a way to do it in hardware @ryanteck @vpetersson for a v2. Does seem like a bit of a pain for a consumer product. Should ideally fail safe - because well people are stupid sometimes I guess. But probably not as trivial as I'm making it sound of course 🤣

Something to consider anyway

ryanteck, 2021-02-03:
I would guess this would be something that has to be designed into the mPCIe module rather than the gateways themselves so could be something that Milan could look into for that?

shawaj, 2021-02-03:
cc @mmilann

ryanteck, 2021-02-03:
I'd say the risk is fairly low and it's been a common thing with any radio technology. 

However we could for production get a little sticker over the power connector, 

[image]

suprnrdy, 2021-02-03:
I'd say it's worth it. We're not selling to a lot of RF educated customers, so best warn them ahead of time. You can put the sticker next to the power plug.

suprnrdy, 2021-02-03:
No need to engineer a solution, just slap a warning sticker on it

ryanteck, 2021-02-03:
Could be above or below yeah, I thought if over then they couldn't pretend to have not seen it.

Slightly smaller too maybe,

shawaj, 2021-02-03:
Yeah - and/or we could just print it on the plastic. But yes - i agree that would be good

commit 90f4ebeb (2021-02-03): Add info that fixes #30 but also adds some other guidelines too.
closed 2021-04-25 by ryanteck