NebraLtd/Helium-Guides #31: Hotspot Setup Instructions - Add details on setting up the helium app

suprnrdy, 2021-02-03:
https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md#configuring-your-nebra-indoor-hotspot

- Elaborate on setting up wallet (borrow from helium) or point to Helium instructions - This will cover questions from new people, and we should remind them to backup their backup phrase here!

ryanteck, 2021-02-03:
I can't see a step by step guide from helium myself and only https://www.helium.com/app and https://intercom.help/heliumnetwork/en/articles/3287804-app-overview

Neither are a super clear step by step guide so I can write a little step by step guide on how to set it up.

I think I'll create the template pages and write it later today.

ryanteck, 2021-04-24:
Resolved by @salmanfarisvp in 11fc5025b83ed05ea409f20e01dd297995709097

closed 2021-04-24 by ryanteck