NebraLtd/Helium-Guides #33: Summary list of FAQs

shawaj, 2021-02-06:
As our list of FAQs is getting longer and longer I wonder if it makes sense to either they and have them in a sort of expanding accordion type setup or to just have a clickable summary list at the top so you can look at the questions without having to scroll through all the answers

@ryanteck @JohnWhiting any thoughts?

ryanteck, 2021-02-21:
I've started to split the FAQs over three pages, keeping the generic one for more generic questions plus the table. Then an induvidual FAQ page for the indoor & outdoor units. This has helped cut it down and make it clearer.

I think if there's an expanding plugin for Docsify that could be good too so will look into it.

shawaj, 2021-04-30:
@ryanteck did we ever try the built in markdown accordion features that they have in github? Can't remember if we did or not now.

https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab

ryanteck, 2021-04-30:
I haven't looked into this yet.

salmanfarisvp, 2021-05-01:
> https://gist.github.com/pierrejoubert73/902cc94d79424356a8d20be2b382e1ab

[image]

[image]

This is working fine.

salmanfarisvp, 2021-05-07:
Added FAQ summary on Indoor and outdoor FAQ. https://github.com/NebraLtd/Helium-Guides/commit/0e973005fcccf4136fd34f5a6a3112d9d86461ee

closed 2021-05-13 by salmanfarisvp