NebraLtd/Helium-Guides #38: PSU specs and datasheet

shawaj, 2021-03-05:
Add a bit more PSU specs like the DC jack output (2.1x5.5x12mm C+) and any other bits that might be useful. Especially for the outdoor.

[Prime Power PS1082 Interchageable UK, Eu, US, Au, 12VDC 1.5A Output 2.1x5.5x12mm C+.pdf](https://github.com/NebraLtd/Helium-Guides/files/6093240/Prime.Power.PS1082.Interchageable.UK.Eu.US.Au.12VDC.1.5A.Output.2.1x5.5x12mm.C%2B.pdf)

https://cdn.shopify.com/s/files/1/0071/2281/3001/files/Prime_Power_PS1082_Interchageable_UK_Eu_US_Au_12VDC_1.5A_Output_2.1x5.5x12mm_C.pdf?v=1614976589

ryanteck, 2021-04-26:
@shawaj Did you want me to wait for the Nebrafied datasheet or upload the ones we currently have until we finish the new datasheets?

salmanfarisvp, 2021-06-05:
can we add the PSU details here https://helium.nebra.com/datasheets, let me know.

salmanfarisvp, 2021-09-20:
✅ Added https://github.com/NebraLtd/Helium-Guides/tree/main/docs/datasheets/accessories

closed 2021-09-20 by salmanfarisvp