NebraLtd/Helium-Guides #44: Port forwarding

shawaj, 2021-04-07:
Re https://discord.com/channels/404106811252408320/796070988118360094/829393946072842340

Possibly add port forwarding info

[image]

vpetersson, 2021-04-08:
You sure this is required? My device seems to be relaying data just fine over just regular NAT.

vpetersson, 2021-04-08:
@ryanteck looks like a screenshot from pfSense :)

Do we have any technical details on this from the Helium folks?

ryanteck, 2021-04-08:
@vpetersson Did a little more looking into it.

It isn't something that is required, as hotspots are still able to communicate with eachother over P2P. But I believe that the network relys on some units being forwarded so the others can communicate via P2P or something.

I think that maybe it could be something we add to the FAQs as a "You may wish to port forward port 44158". But not enough to do a full tutorial as everyone will want a tutorial for their specific network setup which would be too much of a PITA to deal with.

vpetersson, 2021-04-08:
Yeah that's fair. Would be interesting to know what the financial upside is to enable port forwarding.

ryanteck, 2021-04-10:
It looks like now Helium have found too many hotspots aren't port
forwarding so they've added an indicator to display this now.

I think now it's recommended more, it could be worth doing the guides.

We could try and see if with community help we can document as many brands
as possible.

vpetersson, 2021-04-11:
I can confirm that doing a port forwarding on 44158/TCP to the node does remove the "relayed" state that I had briefly previously.

salmanfarisvp, 2021-06-05:
Port forwarding information is here https://helium.nebra.com/security . do we need to add anything more here?

shawaj, 2021-06-05:
@salmanfarisvp I think the idea, initially, was to have it so that we could have community sourced guides for port forwarding on specific routers to help people out. Maybe that is overkill though I'm not sure

salmanfarisvp, 2021-06-05:
That's good, I think we can make one first and use it as a template.

shawaj, 2021-06-05:
Yeah good plan. I can probably do one for my Virgin Media router and for a Sky UK one as well

shawaj, 2021-06-05:
made a folder for screenshots https://github.com/NebraLtd/Helium-Guides/tree/main/docs/media/screenshots/portforwarding

salmanfarisvp, 2021-06-05:
Great, Can we add the docs here https://helium.nebra.com/handy-guides/port-forwarding/overview , I'll add that.

neildotwilliams, 2021-06-05:
Hi. Are you aware of this site?
[Port Forward](https://portforward.com)
More specifically they have screenshots for lots of routers:
[Routers](https://portforward.com/router.htm)
[virgin hub](https://portforward.com/virgin-media/super-hub-2/)

shawaj, 2021-06-05:
@salmanfarisvp created this one as a template https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/port-forwarding/vm-superhub-v3.md

shawaj, 2021-06-05:
@neildotwilliams nice one! hadn't seen that before, but granted I have never searched for instructions on how to do this before. 

@salmanfarisvp I guess we can create a few of our own, but also link to this website as well perhaps, since it already has loads...

shawaj, 2021-06-05:
Just linking this guide from helium, for future reference https://intercom.help/heliumnetwork/en/articles/3207912-troubleshooting-network-connection-issues

goe91, 2021-06-24:
Hello!
If i have 2 devices, how can i forward port 44158? Port forward works with 1 ip..... can someone help me?

shawaj, 2021-06-25:
@goe91 you can't port forward the same port on multiple devices on the same network

closed 2021-07-28 by shawaj

lvivlion, 2021-08-30:
@shawaj I have set up port forwarding on my hotspot. Port seems to be open for a random amount of time 1-2-3-8 hrs then closes. And requires a restart to open back up.      Do you know what closes the port or opens? Can we set up to reopen the port if it's closed? I called ISP they said they are not blocking. Since advanced security feature is disabled.  nebra ticket 259956

commit b9ef42da (2021-07-28): Add more info and links

Add links to info about port forwarding and also to router specific guides

Closes: #44