NebraLtd/Helium-Guides #51: Security Overview

ryanteck, 2021-04-21:
Add content to the security overview page.

ryanteck, 2021-04-21:
Good first issue for @salmanfarisvp?

vpetersson, 2021-04-21:
@ryanteck Do you have any bullet points of content you think we should cover here?

salmanfarisvp, 2021-04-21:
> @ryanteck Do you have any bullet points of content you think we should cover here?

Yeah, that will be very useful.

ryanteck, 2021-04-21:
I'm thinking maybe just a generic overview of what steps have been taken.

Maybe something like

- Uses a lightweight version of linux
- Frequent OTA updates to ensure software is up to date
- Code is split into parts to help keep the unit secure
- Only ports required are open on the unit
- No SSH 

That type of thing, plus maybe a blurb of how it's relatively low risk having a Lora Gateway due to the nature of it?

salmanfarisvp, 2021-04-22:
I found this [Helium Security](https://intercom.help/heliumnetwork/en/articles/3286710-security) docs, do we need to make a generic one like this or need to be specific for the nebra miner?

ryanteck, 2021-04-22:
I think slightly more specific to ours @salmanfarisvp but some good points there which would be worth using as well.

Possibly also adding in content from https://www.balena.io/docs/learn/welcome/security/

We could have a port list a bit like the table at https://www.balena.io/docs/learn/welcome/security/#ports-used

Which would be the 3 ports used by balena, plus then add in the helium miner port.

salmanfarisvp, 2021-04-22:
Thanks for sharing the balena docs, it's looks good. for creating the device-specific thing, I think I should wait for the device to arrive. what do you think?

ryanteck, 2021-04-22:
Hmm I'm not aware of anything in this document that requires an actual device to write it?

closed 2021-04-25 by ryanteck