NebraLtd/Helium-Guides #54: Write antennas / connectors guide

ryanteck, 2021-04-23:
We could do with a little guide just explaining the connectors required on the cables etc.

ryanteck, 2021-04-24:
Just a rough guide that just re-details the connectors on the hotspots and explains the primary methods of connection.

Primarily 3 different possible combinations which are:

Both units - Included antenna screws directly into hotspot.

Then optional / other alternatives are:

Indoor unit - the 3 glass fiber ones we sell can be connected to the indoor unit with an RP-SMA to N Type cable. (Included with antenna)

Outdoor unit - The 3 glass fiber ones we sell can be connected directly to the unit, or with an N Type extension (sold seperately)

salmanfarisvp, 2021-04-26:
Started https://github.com/NebraLtd/Helium-Guides/tree/main/docs/handy-guides/antenna

salmanfarisvp, 2021-04-27:
Added [antenna connectors ](ttps://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/antenna/connectors.md )and [antenna connection docs](https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/antenna/connections.md) @ryanteck please review.

ryanteck, 2021-04-28:
Looking good @salmanfarisvp ,

Only thing would be to add a diagram to show the outdoor hotspt directly connecting to the antenna too.

salmanfarisvp, 2021-04-28:
Sure, Which antenna? The 3dBi one?

ryanteck, 2021-04-28:
Yeah, connects directly into the hotspot as well and via cable.

salmanfarisvp, 2021-04-28:
Ok, I'complete. 

> via cable.

It's already there https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/antenna/connections.md#outdoor-hotspot---glass-fiber-antenna-n-type-connection

KevinWassermann94, 2021-04-29:
Can we add a guide for lightning arrestors aswell?

salmanfarisvp, 2021-04-29:
> Yeah, connects directly into the hotspot as well and via cable.

Done. 

and do we need to put all antenna docs in one markdown file or need to be separate like I did? which on you recommend!

salmanfarisvp, 2021-04-29:
> Can we add a guide for lightning arrestors aswell?

can you share the link?

KevinWassermann94, 2021-04-29:
> > Can we add a guide for lightning arrestors aswell?
> 
> can you share the link?

https://www.thethingsnetwork.org/docs/gateways/lightning-protection/index.html

There should be a lot of useful information

salmanfarisvp, 2021-05-01:
Added a little bit about surge protectors here: https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/surge-protectors.md

If I miss something, please let me know.

salmanfarisvp, 2021-05-01:
> https://www.thethingsnetwork.org/docs/gateways/lightning-protection/index.html

Thanks for sharing, this is useful! I'll try to add this too.

closed 2021-05-03 by ryanteck