NebraLtd/Helium-Guides #55: Certification docs

shawaj, 2021-04-24:
Draft reports...

[2104036 FCC draft report1.zip](https://github.com/NebraLtd/Helium-Guides/files/6368660/2104036.FCC.draft.report1.zip)
[2104034 CE draft report 2.zip](https://github.com/NebraLtd/Helium-Guides/files/6368661/2104034.CE.draft.report.2.zip)

Add final ones when available

shawaj, 2021-04-24:
https://github.com/NebraLtd/Helium-Guides/tree/main/certification

shawaj, 2021-04-25:
[JYTSZE2104034   Nebra Smart Indoor LoRa Gateway 470 Mhz, 868 Mhz, 915Mhz RED.zip](https://github.com/NebraLtd/Helium-Guides/files/6372128/JYTSZE2104034.Nebra.Smart.Indoor.LoRa.Gateway.470.Mhz.868.Mhz.915Mhz.RED.zip)

ryanteck, 2021-04-25:
Just commenting to note why, I've moved the certifications to a subdirectory under the docs folder as this then allows the website to serve the documents.

These can then be found under the certifications URL at https://helium.nebra.com/certifications/

shawaj, 2021-04-25:
@ryanteck nice one! Should it render as a page or just as an index?

ryanteck, 2021-04-25:
Currently just an index as it generates the PDFs on push.

I don't think we need to make a page on the documentation itself with a list of the docs as I've then got it to link to the correct ones from the page on the device itself.

shawaj, 2021-04-27:
https://fccid.io/2AZDM-HNTIN

https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=Pa6qrAs0d%2FUOLfrC8FZExg%3D%3D&fcc_id=2AZDM-HNTIN

shawaj, 2021-04-28:
[JYTSZE2104036 Nebra Smart Indoor LoRa Gateway FCC ID 证书报告.zip](https://github.com/NebraLtd/Helium-Guides/files/6393544/JYTSZE2104036.Nebra.Smart.Indoor.LoRa.Gateway.FCC.ID.zip)

shawaj, 2021-04-28:
Closing for now and will add any new certs directly to repo as necessary

closed 2021-04-28 by shawaj