NebraLtd/Helium-Guides #56: Add details about surge protectors

shawaj, 2021-04-24:
Add details about surge protectors

ryanteck, 2021-04-25:
Primarily I would guess it's writing a little document on surge protecting your unit.

The two main options are:

**Antenna Surge Protector** this goes between the antenna and the hotspot and should be grounded.  This one helps protects the hotspot against damage.

**Ethernet Surge Protector** This goes between the hotspot and your ethernet router / switch. This one would only help protect against other devices getting damaged and not the hotspot itself.

Possibly put that both can be used together. And that for the grounding of the antenna protector should be done in accordance to local regulations (as I suspect this is something where guidance varies based on local electrical codes).

salmanfarisvp, 2021-05-01:
Added a little bit about surge protectors here : https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/surge-protectors.md

If I miss something, please let me know.

closed 2021-05-03 by ryanteck