NebraLtd/Helium-Guides #59: create "certifications" overview page

shawaj, 2021-04-26:
Was thinking we could make a little table with certification info...

Some inspiration here - https://www.raspberrypi.org/documentation/hardware/raspberrypi/conformity.md

There are some standard texts we have to put in the "user guide" for FCC...something similar to what is shown here. Was thinking we could just put it on this same page and link to it.

ryanteck, 2021-04-26:
I did create a table already on the product overview pages with a list of certifications?

[image]

ryanteck, 2021-04-26:
Reason why was so that it then shows under the product's generated datasheets at https://helium.nebra.com/pdfs/indoor-overview.pdf for example.

I can move it into it's own page if you like @shawaj but wasn't sure if you had seen it was already on the products page.

shawaj, 2021-04-27:
FCC Statement
1. This device complies with Part 15 of the FCC Rules. Operation is subject to the following two conditions:
(1) This device may not cause harmful interference.
(2) This device must accept any interference received, including interference that may cause undesired operation.
2. Changes or modifications not expressly approved by the party responsible for compliance could void the user's authority to operate 
the equipment.
NOTE: 
This equipment has been tested and found to comply with the limits for a Class B digital device, pursuant to Part 15 of the FCC Rules. 
These limits are designed to provide reasonable protection against harmful interference in a residential installation.
This equipment generates uses and can radiate radio frequency energy and, if not installed and used in accordance with the 
instructions, may cause harmful interference to radio communications. However, there is no guarantee that interference will not occur 
in a particular installation. If this equipment does cause harmful interference to radio or television reception, which can be determined 
by turning the equipment off and on, the user is encouraged to try to correct the interference by one or more of the following measures:
Reorient or relocate the receiving antenna.
Increase the separation between the equipment and receiver.
Connect the equipment into an outlet on a circuit different from that to which the receiver is connected. 
Consult the dealer or an experienced radio/TV technician for help.

ryanteck, 2021-04-28:
@shawaj I've added the above text for you to the overview pages for the hotspots.

[image]

This then adds it to the generated user guides for you if this is OK?

shawaj, 2021-04-28:
@ryanteck looks great! Nice one

shawaj, 2021-04-28:
@ryanteck agreed on the table on the individual product page, that's a better solution I think.

Maybe we just add links there to the repo/download page for the actual docs?

ryanteck, 2021-05-03:
Links added for the certs.

closed 2021-05-03 by ryanteck