NebraLtd/Helium-Guides #65: IP67 case 2d and 3d models

shawaj, 2021-05-05:
People have asked for these so might as well add them somewhere in guides...

[ip67case.zip](https://github.com/NebraLtd/Helium-Guides/files/6425060/ip67case.zip)

salmanfarisvp, 2021-05-05:
Shall we add on the https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md

shawaj, 2021-05-05:
Yes maybe we can upload to the files to the media directory and then add the links in the dimensions section perhaps

shawaj, 2021-05-05:
On second thought, maybe better to have
Folder for docs->media->cases->outdoor and then indoor as well to keep it better organised. As some people have asked about those files too and at some point was going to open source everything anyway so could be a better way

salmanfarisvp, 2021-05-05:
Ok 👍. **docs->media->cases->outdoor**  .

shawaj, 2021-05-05:
This is the 3d files for light gateway
[Light.zip](https://github.com/NebraLtd/Helium-Guides/files/6425093/Light.zip)

Need to find/get the ones for indoor as well

salmanfarisvp, 2021-05-05:
> On second thought, maybe better to have
> Folder for docs->media->cases->outdoor and then indoor as well to keep it better organised. As some people have asked about those files too and at some point was going to open source everything anyway so could be a better way

[image]

Added the outdoor resource list, having an issue with the link and I'll for that soon.  link: https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md

closed 2021-05-27 by salmanfarisvp