NebraLtd/Helium-Guides #67: Document dashboard

shawaj, 2021-05-08:
Once the features are a bit more set in stone and frequency of changes is lower, we should document the remote management dashboard and add it to the guides https://dashboard.nebra.com

Obviously will need to have some devices first too...

shawaj, 2021-08-18:
@KevinWassermann94 @salmanfarisvp we sold do this at some point soon, as we get a lot of repeated questions about certain things on dashboard

salmanfarisvp, 2021-09-04:
Added dashboard setup and overview at https://github.com/NebraLtd/Helium-Guides/commit/ade7ec049ee020212ecf916f714d53583d85418d . 

@shawaj Please take a look https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/dashboard.md .

closed 2021-09-20 by salmanfarisvp