NebraLtd/Helium-Guides #71: Documentation review - Indoor Hotspot -

fouad-semaan, 2021-05-15:
Bold text should be reformulated/edited in the following files:

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md#preparing-your-nebra-indoor-hotspot

- Step 3: If you are using a wired Ethernet connection, **connect an Ethernet cable between a router or switch and the Ethernet jack on the Hotspot.**
- Step 5: Finally connect the DC Jack from the power supply into the power receptacle on the Hotspot.

You should then see the lower LED on the back of the unit **turn on**, the hotspot will now take up to 5 minutes to configure for it’s first boot. --> **turns on**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md#configuring-your-nebra-indoor-hotspot

Configuring Your Nebra Indoor Hotspot
--------------------------------------------
To configure your Hotspot you will require the Helium Network application installed on a Mobile Phone **and for you to have gone through the account setup process to continue.**

Step 2: Next click Set up **Hotspot , from** here you will want to select Nebra Indoor Hotspot. --> ** Delete space before the comma :P**

Step 8: Finally you can confirm the location of your hotspot. Click continue and you should be presented **with a map to then place where your hotspot is on the app.**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/troubleshooting.md#no-lights-are-turning-on

No lights are turning on:
---------------------------
You should see the lower light on the back of the unit **turn on** as soon as power is applied. If this is the case, ensure the power supply is firmly plugged into the wall outlet, the power jack is plugged into the unit and that the wall outlet is switched on.  --> **turned on**

Only the lower light will turn on **to begin with** until the unit is running. --> At the beginning, Only the lower light....

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/troubleshooting.md#my-hotspot-is-not-showing-in-the-helium-app

My hotspot is not showing in the Helium App:
--------------------------------------------------
**It can sometimes take 1 minute from power on for the Hotspot to show in the App, simply re-scanning a few times can then find the hotspot.**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/troubleshooting.md#updating-your-nebra-hotspot

Updating your Nebra Hotspot
---------------------------------
If after completing initial troubleshooting steps above you're still having issues, **or you are advised to you may find your issues can be resolved by ensuring it is running the latest up to date software.**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/indoor-faq.md#what-antenna-is-included-with-the-hotspot

We are currently confirming the final model of antenna included with the indoor hotspot.  --> **Not confirmed yet?**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/indoor-faq.md#can-i-use-poe-with-the-indoor-hotspot

If you wish to use PoE with the Indoor Hotspot you will require an active **PoE splitter** that outputs 12V to the unit. Alternatively you can use passive PoE adapters as long as the power provided to the unit is between 9 and 16 Volts and can provide 15W of power. --> **Injector then Splitter?**

salmanfarisvp, 2021-05-15:
Thanks, Only need to make the text bold that mentioned paragraphs right?

fouad-semaan, 2021-05-15:
The bold text, except the titles, should be reformulated in English.

ryanteck, 2021-05-15:
@fouad-semaan Could you help me with the following?

> 
> * Step 3: If you are using a wired Ethernet connection, **connect an Ethernet cable between a router or switch and the Ethernet jack on the Hotspot.**

What's incorrect with the above? To me this reads fine.
 
> You should then see the lower LED on the back of the unit **turn on**, the hotspot will now take up to 5 minutes to configure for it’s first boot. --> **turns on**

"Turns On" doesn't sound correct to me, verbally the led on the back of the unit would turn on. Turns on would be past tense?

> ## No lights are turning on:
> You should see the lower light on the back of the unit **turn on** as soon as power is applied. If this is the case, ensure the power supply is firmly plugged into the [...]

As above, turned on would be past tense.

> Only the lower light will turn on **to begin with** until the unit is running. --> At the beginning, only the lower light....
> 

Reworded

> ## Updating your Nebra Hotspot
> If after completing initial troubleshooting steps above you're still having issues, **or you are advised to you may find your issues can be resolved by ensuring it is running the latest up to date software.**

I've re-worded this as then realised that if we're guiding them on what to do they wouldn't be at this page. Reads slightly better now.

> We are currently confirming the final model of antenna included with the indoor hotspot. --> **Not confirmed yet?**

Added in details.

> If you wish to use PoE with the Indoor Hotspot you will require an active **PoE splitter** that outputs 12V to the unit. Alternatively you can use passive PoE adapters as long as the power provided to the unit is between 9 and 16 Volts and can provide 15W of power. --> **Injector then Splitter?**

I think for this just splitter is fine, as it's indicating the hardware required for the hotspot more on a "I know enough about POE but not sure what splitter I need" route rather than a "How do a DIY POE" with it.

fouad-semaan, 2021-05-15:
**connect an Ethernet cable between a router or switch and the Ethernet jack on the Hotspot.**
Ahhh now I understand it, **switch** here is a word and not a verb. So maybe we can add **a switch**. Newbies as me will understand better.

For the rest, that's ok! 

Thanks.

closed 2021-05-21 by salmanfarisvp