NebraLtd/Helium-Guides #72: Outdoor cases not linking properly?

shawaj, 2021-05-15:
@ryanteck the pdf in this folder links correctly but not the other files. 

Any idea why?

https://github.com/NebraLtd/Helium-Guides/tree/main/docs/media/cases/outdoor

You can see the links here:

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md#resources

ryanteck, 2021-05-15:
The server is only configured to serve PDFs as a static file.

I believe the only directories where this does not apply is the certifications and pdfs folders.

I will resolve these now.

However, would it be better to move these to the dedicated Datasheets page? https://helium.nebra.com/datasheets

shawaj, 2021-05-16:
Yes probably a good idea. And then from the "Dimensions" section of the page we could just put a link saying something like "You can also find 2D and 3D drawings of the outdoor IP67 case here." Or something to that effect

salmanfarisvp, 2021-05-21:
my intention was when the user clicks the resource the files should download and work on the local server but not when serving online!

ryanteck, 2021-05-21:
Sorry @salmanfarisvp Forgot to fix this.

I've just fixed the server configuration now.

However could these be moved to the datasheets page instead?

salmanfarisvp, 2021-05-21:
Thanks @ryanteck , Sure I'll move to datasheet page.

salmanfarisvp, 2021-05-21:
Added on datasheet page. 

[image]

closed 2021-05-27 by salmanfarisvp