NebraLtd/Helium-Guides #73: Documentation Review - Outdoor Hotspot -

fouad-semaan, 2021-05-16:
https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md#outdoor-helium-hotspot-overview

## Specifications

| Specification | Nebra Outdoor Hotspot |
| --- | ---  |
| **RRP** | £350 ex VAT |
| **Case** |  |

 **Empty Case above**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md#antenna-specifications

| Specification | 470Mhz Model | 868 & 915Mhz Models |
| --- | --- | --- |
| **Frequency Range** | 420-480 | 860-930 Mhz|
| **Peak Gain** |  | 3 dBi |
| **VSWR** |  | ≤1.5 |
| **Input Impediance** | 50 Ohms | 50 Ohms |
| **Length** |  | 30CM |

**Empty for 470MHz model above**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/overview.md#button

The Nebra **Indoor** Hotspot has a button on the smaller board inside the unit. --> **Outdoor**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/troubleshooting.md#outdoor-hotspot-troubleshooting

# Outdoor Hotspot Troubleshooting

If you are having trouble with setting up your hotspot or got a *query?,* the following tips may help. --> **Delete comma**

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/lte-config.md#required-hardware

* micro SIM card with ~30GB per month data package --> **In the FAQs we have recommended 50GB** 

https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/quick-start.md#preparing-your-nebra-outdoor-hotspot
**If not done, let us create an issue as a reminder to write this section when ready**

salmanfarisvp, 2021-05-21:
Thanks, @fouad-semaan  , Fixed 4 of them and working on adding 470MHz details. 

- [X]  Empty Case above
- [ ] Empty for 470MHz model above
- [X] Outdoor
- [X]  50GB

I don't get the last one https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/quick-start.md#preparing-your-nebra-outdoor-hotspot
If not done already, let us create an issue as a reminder to write this section when ready, can you elaborate pease.

fouad-semaan, 2021-05-21:
https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/quick-start.md#preparing-your-nebra-outdoor-hotspot

This section will be written soon.

The content still empty so let us create an issue as a reminder to write it.  @salmanfarisvp

shawaj, 2021-07-28:
Closing as duplicate of #104

closed 2021-07-28 by shawaj