NebraLtd/Helium-Guides #77: Add information about LED status and colours

bigdaveakers, 2021-05-27:
The photo in the guide shows a green LED on the bottom but actually this is amber. It is worth explaining that the LEDs are actually single colour and the lower one will not turn green! The photo should be updated and the description updated. Should also say that the top one will flash until the boot is complete and go to solid green.

shawaj, 2021-05-31:
@bigdaveakers guessing you mean the second image here? https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md#preparing-your-nebra-indoor-hotspot

shawaj, 2021-06-11:
@KevinWassermann94 or @ryanteck could you take a picture of the correct LED colours for this?

salmanfarisvp, 2021-06-20:
@ryanteck added information here https://helium.nebra.com/handy-guides/advanced-diagnostics?id=external-lights. @shawaj can you check that.

shawaj, 2021-06-21:
@salmanfarisvp the problem is that the LED shows green at the bottom in this picture...

[image]

It should be green at top and yellow bottom

salmanfarisvp, 2021-06-21:
Ok, Got it. Will try to update the image ASAP.

salmanfarisvp, 2021-06-26:
Added hotspot back side image https://github.com/NebraLtd/Helium-Guides/commit/cb897f0ee939db5f9a7a5653f08c90ad5008ddde 

[image]

shawaj, 2021-07-29:
Think this has been fixed for the indoor at https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md

And also on the advanced diagnostics page https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/advanced-diagnostics.md

Although neither are particularly clearly worded compared to how the lights actually function and the detail is missing for the outdoor hotspot

salmanfarisvp, 2021-10-04:
Added more details here https://github.com/NebraLtd/Helium-Guides/blob/main/docs/handy-guides/advanced-diagnostics.md#green-light-blinking--status

closed 2021-10-04 by salmanfarisvp