NebraLtd/Helium-Guides #80: Add information about running WiFi and Ethernet together

edpgcooper, 2021-06-02:
I have both interfaces connected (my device has two IP addresses), is it smart enough to prioritise Ethernet for Helium traffic when it's available?  If not how do I stop it using WiFi without a total reset?

vpetersson, 2021-06-04:
Interesting question. The network is managed by [Network Manager](https://wiki.gnome.org/Projects/NetworkManager) behind the scenes. It is certainly possible to run both WiFi and Ethernet simultaneously, but I'm not sure if Network Manager is intelligent enough to prioritize properly. I know that macOS will automatically route traffic over Ethernet if both Ethernet and WiFi is present. 

This would be relatively easy to test, but I cannot personally allocate the bandwidth for it at the moment.

shawaj, 2021-06-05:
One thing I also wondered, is whether having both simultaneously connected can cause any issues with libp2p successfully communicating with peers...?

edpgcooper, 2021-06-17:
I've actually tried many times to clear the WiFi credentials from the device with zero luck - perhaps some clarification is needed as to how you can remove the WiFi credentials when you just want to use Ethernet

devnetworks, 2021-08-06:
I set up separate policies on my Fortigate firewall for traffic on 44158. One coming from Wi-Fi, the other coming from wired Ethernet. Placed them before all other rules so the counters would reflect traffic. Tested with a Nebra hotspot. I plugged it in, cleared all logs, and plugged the Ethernet cable in shortly after. Pay attention to "Bytes" at the very end of the image. 

[image]

Couple minutes later:

[image]

From a different view. Wifi Hotspot Traffic should appear.

[image]

Now I will unplug the Ethernet cable.

[image]

[image]

Back to wired.

[image]

I was not seeing traffic from the wired and had no ethernet lights, so I rebooted. 

[image]

Suspect it was a switch security setting that caused the port not to light up, so trying again. Switching back to wireless in the next picture.

[image]

Wired comes back, and the sessions have dropped from Wifi.

[image]

Port stayed online this time. As seen above, I wasn't seeing much traffic on Wired. I reset the ipv4 policy counters:

[image]

[image]

DGMurdockIII, 2021-09-07:
I would love easy way to remove the wifi from being listed in my router