NebraLtd/Helium-Guides #81: Security page open source software section

shawaj, 2021-06-03:
The column for "description" here doesn't seem to make a lot of sense?

[image]

salmanfarisvp, 2021-06-05:
Fix description  https://github.com/NebraLtd/Helium-Guides/commit/6cd494e0cfb3e3d667f46d130608a6ec2e724ccf , https://github.com/NebraLtd/hm-gwmfr description is missing, I'll add that after checking with @ryanteck .

[image]

shawaj, 2021-06-05:
Gwmfr is for provisioning the ECC secure element chip, it's only used during manufacture essentially

salmanfarisvp, 2021-06-05:
Thanks @shawaj , added that too, now it's complete, please check and close 🙂👍

[image]

shawaj, 2021-06-05:
Looks good to me @salmanfarisvp

closed 2021-06-05 by shawaj