NebraLtd/Helium-Guides #91: Improve uptime

shawaj, 2021-07-01:
The server is proving a bit unreliable based on the status. https://status.nebra.com/

It's already proxies by cloudflare but we should improve this if possible

vpetersson, 2021-07-02:
I'd look into moving the hosting elsewhere. Probably easiest is we just use GitHub Pages to host it. Shouldn't be too difficult as it's just a static page as far as I can tell.

shawaj, 2021-07-02:
@vpetersson we had it on GitHub pages before but it didn't work correctly. That's why we moved it

vpetersson, 2021-07-02:
Roget that. Need to look into it in more details to fully understand the requirements.

vpetersson, 2021-11-02:
Closed in favor of https://github.com/NebraLtd/Helium-Guides/issues/225

closed 2021-11-02 by vpetersson

shawaj, 2021-11-04:
Just FYI, the downtime previously was caused by it rebuilding the site periodically and not being cached. Just for future reference