NebraLtd/Helium-Guides #95: Guide for switching concentrator / region

shawaj, 2021-07-16:
Specifically in relation to https://github.com/NebraLtd/hm-pktfwd/issues/18

iHextic, 2021-11-20:
Do you guys know any concentrators compatible with the outdoor version

shawaj, 2021-11-21:
> Do you guys know any concentrators compatible with the outdoor version

Just use the one it came with unless instructed otherwise by our support team

serbyxp, 2021-12-19:
Switched out broken SX1301 for new SX1302. I updated app.json under env service pkt-fwrdr, “VARIANT: “CONCENTRATOR_RESET_PIN=25” as it is the only different pin for US915 global.conf…
It did not work. I also attempted to add 

“VARIANT”: “1302_LORA_PKT_FWD_DIR” 
“VARIANT”: “1302_REGION_CONFIG_DIR”

serbyxp, 2021-12-21:
Im not sure but I think, the GL1571-UX that came with my Nebra miner, I cant really understand the documentation part where it talks about the different versions USB / SPI ,but the MAXIIOT documentation I found refers to  a SPI / USB "HOST" device, being pull up and the other pull down, so I believe its a GPIO setting that may need to be adjusted. Ive been at it for like 4 days but I have no idea about this balenaos balena-engine, and its on your end that it can be easily adjusted through the fleet cloud service, not sure if there is a work around for this issue. * with out me having to recompile anything. I dont know enough about the ENV variables and how to adjust them without messing something up.