NebraLtd/Helium-Guides #96: Add Video Guides

KevinWassermann94, 2021-07-16:


shawaj, 2021-07-16:
@KevinWassermann94 FYI have sent you and the others indoor and outdoor miners each to help with debugging etc and for stuff like this

shawaj, 2021-07-16:
@KevinWassermann94 let me know if any other stuff would be useful like the LTE module etc?

KevinWassermann94, 2021-07-16:
@shawaj Sure, I will use these for creating the videos.
LTE module would be great as I could create the video guide for that in the same run

shawaj, 2021-07-16:
Cool. Let me or Jacqui know if you need anything else 👍

KevinWassermann94, 2021-07-16:
This is what I am planning:

Indoor:
- Reseating daughterboard
- Reseating compute module
- Reseating Lora concentrator
- Fixing antenna port

Outdoor:
- Reseating daughterboard
- Reseating compute module
- Reseating Lora concentrator
- Installing LTE module
- Installing antennas (Wifi, LTE, Lora)

Btw. is the 8dbi antenna too big for direct mounting? If yes, we should have that information somewhere.

Any additions?

shawaj, 2021-07-16:
@KevinWassermann94 you mean direct mounting to the outdoor right? If so, no it's fine. All of our test units of outdoor have 8dbi attached

shawaj, 2021-07-16:
@KevinWassermann94 do you have an 8dbi? Would it be helpful to send one?

KevinWassermann94, 2021-07-16:
@shawaj Yes I meant direct mounting. Thanks for the info
I have a 5.8dbi and 8dbi antenna that I can use

shawaj, 2021-07-16:
Perfect 👍 expect lots of parcels next week 😉

shawaj, 2021-07-16:
Have you got a decent camera rig or you need some stuff? We have some stuff at the warehouse if not

KevinWassermann94, 2021-07-16:
Sounds great! 
I have a Canon EOS 700D and a tripod so I should be good camera wise. For optimal lighting I will improvise a bit

shawaj, 2021-07-16:
Cool, if you need any lighting gear let us know!

salmanfarisvp, 2021-07-20:
mine is also on the way and I'll do the step by step text-based documentation.

ChristopherRush, 2022-04-18:
@KevinWassermann94 Do we still need to work on this?

KevinWassermann94, 2022-04-18:
@ChristopherRush All good here

closed 2022-04-18 by KevinWassermann94