NebraLtd/Helium-Guides #98: All 4 Outdoor Nebra miners dead

Kayz86, 2021-07-18:
Hi I just received 4 outdoor miners yesterday 17/07/21 (Batch 2) here in the UK.

It appears none of them are powering up at all. Even after connecting them via ethernet my router is not picking them as even a connection type.

- I have the antennas connected
- I have checked the inside and all parts and components are plugged in, nothing is loose.
- I have connected all 4 via PoE ethernet to my router
- I have tested all cables with a cable tester all cables working
- I have swapped all cables with other cables
- I have restarted router
- I have pressed the red button for 15 seconds on onto devices to reconfigure bluetooth - that didn't work.
- I have take out the ethernet and have connected via DC mains 1.5v via the motherboard on the inside of the miner again theres no power even after pressing the red button.
- Theres no light anywhere across all 4 devices on or inside the motherboard.
- I have left all 4 plugged in for over 4 hours. No sign of life no bluetooth update or configurations showing.

As I mentioned earlier, before I even start to configure these devices my router should pick up some form of connection I.e. a wire has been plugged into Lan 1 or Lan 2 etc it should at least indicate to me something is connected - but no.

Unfortunately nothing appears to work.

Am I missing something or doing something wrong? Surely all 4 devices cant be defective?

You're help would be much appreciated.

shawaj, 2021-07-18:
What are you using for PoE injector? It's almost impossible they are faulty as they are tested as working out of the factory.

FYI for powering by DC Jack you have to change the jumper near the Ethernet Jack to the other setting

Kayz86, 2021-07-18:
Hi Aaron

Yes I think I am doing something wrong.

I have simply plugged in an ethernet cable into the device and the other end into the router. Not using any injectors, presumably that's how it works? Or am I required to do something else?

I then took out the ethernet cable and inserted the 1.5v DC. Where is the jumper I need to change? I'm familiar with primary and slave drive jumpers presumably it's similar?

See attached photo.

[image]

Kayz86, 2021-07-18:
I see the JP1 (1-2-3) JP3 (1-2) JP4 (1-2-3)

JP1 is on 1 and 2
JP3 is on 1 and 2
JP4 is on 2

I'd be interested to know how to get PoE working with the ethernet cable.

Thank you.

shawaj, 2021-07-18:
@Kayz86 yep you need a PoE injector.

Something like TP link tlpoe 150s or 160s.

Or you could get a multi port switch like TP link tlsf1005p V2

shawaj, 2021-07-18:
The jumper to change for DC / PoE power is this one...

[image]

1-2 for PoE
2 to 3 for DC

Kayz86, 2021-07-18:
Lord help me man. Spent 6 months planning and I'm missing a poe injector!!!? Somebody shoot me now!

I have CCTV plug and play poe I assumed it's the same, clearly not.

Thanks Aaron you may delete this thread.

shawaj, 2021-07-18:
Take a look here also https://helium.nebra.com/handy-guides/poe-explained

And let us know if this doesn't fix the issue and you still need more assistance 👍

closed 2021-07-18 by shawaj