NebraLtd/Helium-Guides #99: Sierra wireless mc7700 mPCIe 4g module

shawaj, 2021-07-18:
Testing this mPCIe 4g module with @bigdaveakers

bigdaveakers, 2021-07-21:
Updated apn details, good to test!

bigdaveakers, 2021-07-25:
Does the USB ID need adding to 

https://github.com/NebraLtd/hm-diag/blob/master/diagnostics-program/hardware_detector.py

Sierra Wireless MC7700 | Needs Testing | Micro Sim | **_1199:68a2_** | No longer manufactured

bigdaveakers, 2021-08-01:
After a couple of failed attempts and resorting to the Quectel module I have now got the Sierra Wireless MC7700 working!

Interestingly after many hours of running it is showing NOT RELAYED still where the Quectel module almost immediately showed RELAYED.

In order to get it running the cellular file needed changing for THREE by removing completely the lines with username and password in. The guide shows that they dont need to be filled in but doesn't show that they should be removed.

shawaj, 2021-08-20:
Thanks for the intel here @bigdaveakers will update that

shawaj, 2021-08-20:
Related to https://github.com/NebraLtd/hm-diag/issues/88