NebraLtd/Helium-Guides #100: What are these additional parts

Kayz86, 2021-07-20:
Hi all

So I noticed there are some additional parts that come with the Nebra Miner. I for one cannot figure out how to make use of these extras?

I thought I'd be able to use one to cover up the 4g female antenna but none of the parts fit.

The black parts except for one the rest what do I do with it?

I also noticed the ethernet area where it connects is still exposed to the elements. Is there a way to cover this up after connecting it?

Heres an attached image.

[image]

shawaj, 2021-07-20:
the string thing is part of the sealing mechanism for the case

the white antenna is the wifi antenna, it goes on the connector on the bottom right of the nebra logo 

the little black connector is an M10 cable gland
the medium black connector is an M20 cable gland.
the largest black connector is to cover the external ethernet port and make it waterproof

the three metal plugs are just spares

@KevinWassermann94 @salmanfarisvp when you get your outdoor miners perhaps we should overhaul the "outdoor quick start" page to include more info about contents, how to put the miner together, making sure all bolts and connectors are tight etc to maintain weatherproofing etc etc.

Kayz86, 2021-07-20:
Thanks for the speedy response.

The smallest and medium glands presumably we cant use them now as they dont fit anywhere?

Yes the largest one fits the ethernet area, presumably after connecting the ethernet cable theres no way to put that large gland on to cover it up? I wanted to further waterproof it and have taped the bottom area with duct tape.

The white antenna, I notice its attached to the simcard area. Is the antenna required for the simcard to work or can we do without it as were not using wifi and instead the simcard?

Kayz86, 2021-07-20:
Thank you.

Are the smallest and medium glands meant to go in place of the bolts at the top/bottom if I were to take them out?

I will try threading the ethernet through that largest glad again. Last time I tried nothing was passing through.

I thought the white antenna may be for the 4g network. If its solely for the wifi which I'm not using then I'll take it off.

smallest and medium glands

KevinWassermann94, 2021-07-20:
Yes, you will need to remove the bolts to use the glands at these spots.
There are no LTE antennas included with the outdoor miner itself. You will receive 2 LTE antennas when you order the Quectel 4g modem from us. The included antenna is solely meant for Wifi.

Kayz86, 2021-07-20:
> Yes, you will need to remove the bolts to use the glands at these spots.
> There are no LTE antennas included with the outdoor miner itself. You will receive 2 LTE antennas when you order the Quectel 4g modem from us. The included antenna is solely meant for Wifi.

Hi Kevin with all the Nebra units, I received a white antenna.

With the Quectel 4g modem card I also received x2 white antennas, they look exactly the same as the ones I received in the box by default.

I used the white antenna that came in the box for the 4g or am I meant to use the specified white antennas that came with the 4g card? Good thing I've kept them separate as otherwise I dont think I would have been able to differentiate.

shawaj, 2021-07-20:
They are all the same. They are wide band 4g / WiFi antennas

Kayz86, 2021-07-20:
> For the Ethernet cable through the gland cover you need to remove the cover, then remove the rubber gasket and put it around the rubber
> 
> See pictures...
> 
> [image]
> [image]
> [image]
> [image]

Many thanks for this detailed overview.

Thank you

Kayz86, 2021-07-20:
> They are all the same. They are wide band 4g / WiFi antennas

I see thank you. I have attached one at the bottom for the 4g anyway.

Hopefully once I get some time I will attempt the APN and get the 4g working properly.

shawaj, 2021-07-20:
The one at the bottom doesn't connect to the 4g

Can you send some pictures?

jameses986, 2021-07-26:
> the string thing is part of the sealing mechanism for the case
>
> the white antenna is the wifi antenna, it goes on the connector on the bottom right of the nebra logo
>
> the little black connector is [...]

The information about the white antenna is very helpful!  based on the information I read on quick start guide, it states that all antennas should be connected to the miner when it's powered on.  can the Wi-Fi antenna be disconnected while powered on?
thank you.

ChristopherRush, 2022-05-16:
I believe all parts supplied with the outdoor miner is now well documented here - https://helium.nebra.com/#/outdoor-hotspot/overview?id=package-contents