NebraLtd/Helium-Guides #102: Move guides to GitHub Pages

vpetersson, 2021-07-20:
Right now these are served on a remote server. Let's bring them into GitHub Pages instead.

shawaj, 2021-07-20:
We can but it fucks up the URL mapping and PDF generation etc

shawaj, 2021-07-20:
IIRC

vpetersson, 2021-07-20:
For references https://github.com/docsifyjs/docsify/issues/1479

So perhaps we just move to another static page generator that works on GitHub Pages.

shawaj, 2021-07-20:
Related to #101 also

FWIW this does work with GitHub pages you just will have a # in the URL

salmanfarisvp, 2021-07-21:
Can we use the same method that we used previously?

vpetersson, 2021-07-21:
@salmanfarisvp nothing will be changed with regards to the process. Should be fully seamless.

shawaj, 2021-07-21:
@vpetersson https://nebraltd.github.io/Helium-Guides/#/

This is just a demo using the `pages` branch