NebraLtd/Helium-Guides #103: 4G Sim - Nebra Outdoor Miner - no data being used.

Kayz86, 2021-07-20:
Hi so i had updated my miner via ethernet before updating location or antenna via helium up. The device is synced so I know its working.

Now I've moved the mine into an area with no internet access, I have inserted a 4g/5g sim card and it the mine is powered on. However it has been 24 hours now and having checked my online account no data has been used yet.

Is there anything more I need to do? Change jumpers to enable 4g/5g sim card?

Look forward to your valuable assistance as always.

Kind Regards

KevinWassermann94, 2021-07-20:
Hi Kayz86,
You will need to configure the APN settings on your hotspot. 
@shawaj is working on a guide and will post it shortly

Kayz86, 2021-07-20:
Could you explain briefly what APN is or what I need to do. Can I do it remotely or do I have to go to the location and do something on the motherboard.

Really need to get this working now I'm afraid....

shawaj, 2021-07-20:
@KevinWassermann94 @Kayz86 guide is half done...

https://helium.nebra.com/outdoor-hotspot/lte-config

Kayz86, 2021-07-20:
> @KevinWassermann94 @Kayz86 guide is half done...
> 
> https://helium.nebra.com/outdoor-hotspot/lte-config

Thank you.

Can I also ask whether a normal phone sim that has a telephone number attached to it with minutes/texts and data (50gb+) would work in the outdoor miner or does it have to specifically be a data sim?

Thank you

salmanfarisvp, 2021-07-20:
@Kayz86 Yes, a normal SIM will work.

Kayz86, 2021-07-20:
> @Kayz86 Yes, a normal SIM will work.

Thank you Salman, I am now cancelling my Vodafone and settling for a Lyca Mobile Sim card.

KevinWassermann94, 2021-07-21:
@shawaj It seems like you need to install RPi boot for some windows devices. A customer couldn't find the mass storage. I advised him to install it and it has worked after that 
Download: https://github.com/raspberrypi/usbboot/raw/master/win32/rpiboot_setup.exe
1.Download and run the Windows installer to install the drivers and boot tool.
2.Plug your host PC USB into the CMIO USB SLAVE port, making sure J4 is set to the EN position.
3. Apply power to the CMIO board; Windows should now find the hardware and install the driver.
4. Once the driver installation is complete, run the RPiBoot.exe tool that was previously installed.
5. After a few seconds, the Compute Module eMMC will pop up under Windows as a disk (USB mass storage device).

shawaj, 2021-07-21:
@KevinWassermann94 ah yes, @bigdaveakers mentioned this to me yesterday. I only tested it on macOS so didn't realise. Will add to the guide later

Kayz86, 2021-07-21:
Hi all

I made a separate thread about N Type to U.FL Bulkhead antennas and to understand where to fit them. I understand the guide is still being worked on.

Presumably even after we have 4g setup we would still need to get the N Type U.FL Bulkhead installed for it all to work?

I feel it would be much quicker if I just purchase a mifi device and put a sim into that which will give me wifi and then connect the mine to that via wifi?

Jcamaya286, 2022-01-28:
Can I use a sim from sprint wireless to use with 4g LTE adapter?

shawaj, 2022-01-28:
> Can I use a sim from sprint wireless to use with 4g LTE adapter?

@Jcamaya286 you should be able to yes

Jcamaya286, 2022-01-29:
Been waiting for my Outdoor miner for over a year now and just recently got confirmation of tracking #. Have not been doing any reserch or reading until now. I have the 4g LTE adapter but just recently found out that it needs an active sim card. 
Does that mean with an actual phone number?
What happens if it receives a call?
Any help or any other info I might need is greatly appreciated.

salmanfarisvp, 2022-01-29:
@Jcamaya286 Yes, you need an active sim card with a data plan, and nothing happens when a call is received since the hotspot only uses the data. 

You can find more information here at https://helium.nebra.com/outdoor-hotspot/lte-config

closed 2022-01-29 by salmanfarisvp