NebraLtd/Helium-Guides #104: Outdoor documentation To-Do

KevinWassermann94, 2021-07-20:
@salmanfarisvp Feel free to add to this list
- Initial setup (PoE - DC switch etc., describe PoE and make sure there is power on the Ethernet)
- Package contents
- Check specifications
- LED description
- Switch description
- Video guides (I will do these)
- Antenna placement on the hotspot
- Add info that Lora module 2 is supposed to be empty
- Set APN config

shawaj, 2021-07-20:
Added to the above

shawaj, 2021-07-20:
FYI I updated some onboarding stuff for indoor and outdoor on the app https://github.com/helium/hotspot-app/pull/812

And corrected the antenna default for outdoor https://github.com/helium/hotspot-app/pull/807

salmanfarisvp, 2021-07-21:
Added bit outdoor unit hardware details https://github.com/NebraLtd/Helium-Guides/commit/8a263730b97cf7937eb8122b0ea0d69bda12f4d1 , copied from https://github.com/NebraLtd/Smart-LoRa-Gateway

shawaj, 2021-07-21:
@salmanfarisvp good find. I forgot about that info we had there 👍

KevinWassermann94, 2021-07-26:
@shawaj What is the string included with the Outdoor hotspot used for?

shawaj, 2021-07-26:
It's a seal for the other side of the case...

[image]

KevinWassermann94, 2021-07-26:
Ok this will be important for the preperation steps. I will add it together with the other checks for waterproofing the hotspot

system-zero37, 2022-01-21:
Can you add documentation on flashing the SD card with a fresh img, the SD cards have a lifespan of about 2 weeks and support doesn't email back in a timely manner..

vpetersson, 2022-01-21:
We don't use SD cards in any of our units.