NebraLtd/Helium-Guides #106: Connecting to RPi in Windows 10 for LTE setup

bigdaveakers, 2021-07-21:
The guide needs to add a section on installing RPi Boot on Windows 10 to allow connection to the Compute Module

It should be downloaded from here

https://github.com/raspberrypi/usbboot/raw/master/win32/rpiboot_setup.exe

After the initial setup you need to run RPiBoot AGAIN to communicate with the board. The first run simply installs the drivers

shawaj, 2021-07-22:
Related to #103

Tadejspacapan, 2022-02-14:
I have a problem connecting the Nebra outdoor miner via the Rpiboot program. The program displays:
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711...
Loading embedded: bootcode.bin
Failed to claim interface
Loading embedded: bootcode.bin
Failed to claim interface
Loading embedded: bootcode.bin

I tried with different USB ports on my computer, but with no solution. On another Nebra outdoor miner it work perfect also LTE modul.

Do you have any solution?
Thanks