NebraLtd/Helium-Guides #107: Powering up the Outdoor Miner

theSolTrain, 2021-07-22:
Hi

I have just received the Outdoor Miner (UK) and it arrived with several loose parts. I also noticed that the USB Bluetooth and WIFI adaptors were plugged in the wrong ports. I've switched them around and reconnected all of the components. 

I've fired it up with POE however all i'm getting is a single red LED on the mainboard and some heat coming from the Ethernet port on the mainboard. 

When using the Helium app it can't detect it over Bluetooth. 

Any advise would be greatly appreciated.

Cheers
[image]

shawaj, 2021-07-22:
The orientation of the Bluetooth and WiFi dongles does not make any difference whatsoever.

Is it possible for you to get some pictures from a few different angles, and less blurry?

shawaj, 2021-07-28:
@theSolTrain any update on this?

shawaj, 2021-08-12:
@theSolTrain any news here? Will close if we don't hear back from you

shawaj, 2021-08-20:
Closing as no reply

closed 2021-08-20 by shawaj