NebraLtd/Helium-Guides #109: Cannot Port forward on Nebra Outdoor 4g Sim

Kayz86, 2021-07-22:
Hi guys in a very tricky situation. I have discovered ports cannot be opened on simcard 4g Outdoor Nebra devices.

Once the simcard is inside the Outdoor Nebra device the port on the simcard cannot be opened.

I have purchased a wifi mifi device with a 4g simcard inside. I have the Nebra Outdoor device connect to it via ethernet / wifi.

I have previously opened ports on my personal router successfully behind a normal ISP landline into my home.

However i have not been successful in opening port on the 4g router which I have the Nebra connect to.

I took the router to a network specialist to help me open the port or find out why it was not working. After almost 5 hours of investigation we found the problem.

The simcard has a fixed private gateway address 10.0.0.1 which is preventing us from opening ports. Even if we opened it on the router it would not work as the sim has a private gateway.

Does this now mean even if we were to put a sim inside the Nebra Outdoor miner directly  it will not have the said 44158 port open at anytime?

shawaj, 2021-07-24:
The only way to have the port open on a 4g connection is if you have paid extra for a static IP. Otherwise you are most likely behind CGNAT

Kayz86, 2021-07-24:
Correct that's what the network specialist has advised. However there are no 4g providers that can offer static on the sim. They will advise you to purchase their business broadband package.

I have plusnet and have static ip and at another location I have the same.

Just thought I let you know about 4g sims as it's impossible without upgrading to a business package with static IP.

vpetersson, 2021-07-25:
I don't think we can conclude that this is true generally, but might be in some region. It is likely more common that you get an IPv6 IP (or IPs rather) on cellular connections. Unfortunately that isn't very useful as Helium (to my knowledge) doesn't support IPv6 yet.

Kayz86, 2021-07-25:
I am not sure what part you are referring to as not entirely true.

4g mobile networks do not have the option nor do they give you the option to open ports or have ports open. This has been the case for a very long time and continues to be the case today.

I can only speak of networks i have spoken to and have researched upon that are based in the UK

The biggest and largest provider EE does not allow it, nor does Vodafone and nor does one of the cheapest Lyca Mobile.

This leads me to believe none of the UK based networks hav

vpetersson, 2021-07-25:
I can assure you that it is technically possible to get a public (IPv4) IP on a GSM/4G network. It's merely a decision from the carrier's side. Most of them use CGNAT (as @shawaj correctly points out) merely because it is cheaper and that there are a finite IPv4 IPs left out there (which are getting more expensive by the year).

You might have better luck with a virtual carrier that focuses on IoT (such as [Soracom](https://www.soracom.io)).

shawaj, 2021-07-25:
Yeah there are also business SIM plans from O2 and other providers that allow you to have static IPs.

It's just not that common / not possible with "consumer" plans

facilitator999, 2021-08-03:
How much damage is done by the connection being relayed? Is it a big deal breaker?

Almanaka, 2021-08-11:
@Kayz86 are you able to get a connection using the 4G SIM? Or are the CGNAT blocking your connection completely? Trying to my own 4G to work but cannot get any connection. So I am trying to determine if anyone has gotten the 4G to work. Also even though the connection might be relayed.

shawaj, 2021-08-20:
@facilitator999 it can just make things a bit slow sometimes due to having to connect through a peer rather than directly.

However, this week improve massively once everything moves to light hotspots towards the end of the year

Kayz86, 2021-08-22:
> @Kayz86 are you able to get a connection using the 4G SIM? Or are the CGNAT blocking your connection completely? Trying to my own 4G to work but cannot get any connection. So I am trying to determine if anyone has gotten the 4G to work. Also even though the connection might be relayed.

@Almanaka 

I couldn't get it to work and I didn't want to wait, so what I did was I purchased a mini portable router that takes a simcard. I fired that up and then connected the Nebra device to the sim router (mifi) and got it to work like that.

However see my original thread over at https://github.com/NebraLtd/Helium-Guides/issues/103

and the guide for setting up 4g/LTE is here: https://helium.nebra.com/outdoor-hotspot/lte-config

Airminer1847, 2021-10-12:
I still have issues with the 4g lte module after installing the apn. In the Ip diagnosis center it shows as un connected still if anyone knows a fourm for this or answer?

You should look into a raspberry pi VPN, I believe it should bypass the mifi device and get you your port open aswell.

siov4279, 2022-04-22:
My Nebra is having trouble syncing. its been up about 4 days and nothing has been happening?