NebraLtd/Helium-Guides #110: The device does not exist or code is incorrect

sajmi, 2021-07-23:
It is not possible to add nebra indoor to https://dashboard.nebra.com/. After scanning the QR code I get an error: "The device does not exist or the code is incorrect"
How to add nebra indoor to the dashboard?
Thank you

salmanfarisvp, 2021-07-23:
Hi @sajmi , Can you please share the below details to sales@nebar.com , after that please share the ticket number here, we will try to solve the issue as soon as possible. 

```
Distributor:
Order number:
Model of unit:
Mac address of the unit(Shown as ETH on sticker):
Frequency of the unit(Shown as Freq on the sticker):
How are you connecting it to the internet(Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, PoE, Third-party adaptor)
If the issue relates to initial setup of the hotspot:
What make & model of phone are you using?
What version of the Helium App are you using?
Do you have any screenshots of any error codes? 
```

sajmi, 2021-07-23:
ticket already created . Here you are ticket number : Ticket (#256928)

shawaj, 2021-07-25:
@salmanfarisvp FYI you had the email as nebar.com - have updated above...

sajmi, 2021-07-25:
@salmanfarisvp  some update about my problem please ?

salmanfarisvp, 2021-07-26:
Thanks @shawaj . 

@sajmi  Thanks for sharing the information via mail and please keep it connected with the internet, we will be performing a fix and let you know once it is ready over mail.

sajmi, 2021-07-29:
@shawaj @salmanfarisvp  I have received email that my ticket was closed but the problem still  persists. Could you look at that ? 
Thank you

sajmi, 2021-08-03:
I have a serious problem that it is not possible to add it, can you solve it? I've been waiting for almost 2 weeks and I have no answer.
all devices are online all the time

SkyThommy, 2021-08-04:
@shawaj @salmanfarisvp  Guys, lets be honest, your dashboard is not working at all, i have multiple HS from different "batches" and cant register a single one, also non of my friends can do it, so where is the problem actually?

macik08, 2021-08-06:
Hi Nebra Team, 
have same issue. Sent mail with all the info to sales@nebra.com
ticket nr. #260611

LEt me know what to do.

BR

Miroslav

salmanfarisvp, 2021-08-11:
@macik08 @sajmi We are working on the fix right now and we will notify you over the mail once it's fixed.

closed 2021-08-11 by salmanfarisvp