NebraLtd/Helium-Guides #115: Improve all pictures where possible

shawaj, 2021-07-26:
Improve all box contents pictures, perhaps with pictures from Louis or similar @JohnWhiting 

For example, the first picture on this page https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md

KevinWassermann94, 2021-07-30:
@salmanfarisvp Feel free to use any pictures that I have uploaded in the Google Drive as well

salmanfarisvp, 2021-08-01:
@shawaj @KevinWassermann94 updated I[ndoor overview](https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/overview.md) and [quickstart guide](https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-hotspot/quick-start.md) . Please check and let me know your feedback.

KevinWassermann94, 2021-08-03:
@salmanfarisvp Looking nice!

salmanfarisvp, 2021-08-24:
Updated [Outdoor quick-start guide ](https://github.com/NebraLtd/Helium-Guides/blob/main/docs/outdoor-hotspot/quick-start.md)

closed 2021-08-24 by salmanfarisvp