NebraLtd/Helium-Guides #118: Add info about local diagnostics / app diagnostics

shawaj, 2021-07-28:
The advanced diagnostics / troubleshooting / quick start / first days guides make no mention of the local diagnostics page or the helium app diagnostics.

As these are both very useful tools we should document them more thoroughly... How to access them, what information they show etc and link to the info from anywhere relevant

KevinWassermann94, 2021-07-28:
I will do that

shawaj, 2021-07-28:
Related to NebraLtd/hm-diag#64

KevinWassermann94, 2021-07-29:
Added info about in-app diagnostics and ip diagnostics for the indoor hotspot in https://github.com/NebraLtd/Helium-Guides/pull/119

salmanfarisvp, 2021-08-11:
Thanks @KevinWassermann94 , I added a few styling too https://github.com/NebraLtd/Helium-Guides/commit/e041868a915fff58e33a13c8f49b16faeadcd238 .

closed 2021-08-11 by salmanfarisvp