NebraLtd/Helium-Guides #123: No Onboarding Key Found

Hbit2288, 2021-07-31:
When I try to sync my Nebra miner to the Helium app, it picks up the miner but then give me an error code, no onboarding key found? How do I fix this issue? TIA

shawaj, 2021-08-01:
@Hbit2288 can you please email sales@nebra.com and include the following information:

Model of unit?
Mac address of the unit (Shown as ETH on sticker)?
Frequency of the unit (Shown as Freq on sticker)?
How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular)?
How are you powering the unit? (included adaptor, PoE, third party adaptor)

If the issue relates to initial setup of the hotspot, please also include:

What make & model of phone are you using?
What version of the Helium App are you using?
Do you have any screenshots of any error codes?

closed 2021-08-02 by salmanfarisvp

Hbit2288, 2021-08-03:
Hello Aaron,

I sent a support ticket number 258847 on 7/31 with all the requested
information to ***@***.*** Is there anyone I can contact or can you
help me with the support ticket?

Thank You

salmanfarisvp, 2021-08-13:
@Hbit2288 Please try to set up the device and let us know.

Nwilati, 2021-08-28:
I have the same issue , no key found onboarding server , I raised a ticket 3 days ago ticket number 269571 for 2 hotspots having the same issue , please help me accelerate the fix , much appreciated. Thank you

ViniSew, 2021-10-09:
Hi,

Has this issue been resolved as i have the same error message :(