NebraLtd/Helium-Guides #127: CE certification for outdoor hotspots should be uploaded to github

flolege, 2021-08-04:
As title says, the CE certification documentation papers for the outdoor hotspots should be made available on github, so vendors can finally also ship the outdoor miners.

flolege, 2021-08-11:
Please prioritize!

PhilippStummer, 2021-08-20:
@shawaj is there a real chance that there will be CE certs for Europe? I just want to know if I should still be expectant to the outdoor miners from my order from distrelec.

shawaj, 2021-08-20:
Here you go https://github.com/NebraLtd/Helium-Guides/tree/main/docs/certifications/outdoor/CE

closed 2021-08-20 by shawaj

flolege, 2021-08-20:
@shawaj This link does not contain the EU Declaration of Conformity. Could you please issue the EU Declaration of Conformit, we are waiting since weeks. This is mandatory for Distrelec.

PhilippStummer, 2021-08-25:
did you sent it to the vendors like distrelec? They didn't receive anything, so they are still not able to send the outdoor units @shawaj