NebraLtd/Helium-Guides #129: Miner Is Still Loading

Vasilev87, 2021-08-05:
Hi Team,

I received my miners couple of days ago, they are now sycned with the blockchain but I've noticed in the Diagnostics Information that Sycn percentage change from 100% to "Miner Is Still Loading" when the Height Status appear to be higher than the blockchain blocks. 

First screenshot is made few minutes before the second one.

I also asked in Discord and few more people had the same problem.
Can you please provide any information how does this apply on the mining and what further steps (if any) are required from us in order this to be resolved? Or if this is software problem, when we can expect any fix for it?

[image]
[image]

Best regards, Hristo.

KevinWassermann94, 2021-08-05:
The height status reported by the Helium API is not up to date and causes the "Miner is still Loading" to show. Your hotspot is fully operational while this is showing and it has no negative impact. It is just a display error. We are working on a fix here already: https://github.com/NebraLtd/hm-diag/issues/80

Vasilev87, 2021-08-05:
> The height status reported by the Helium API is not up to date and causes the "Miner is still Loading" to show. Your hotspot is fully operational while this is showing and it has no negative impact. It is just a display error. We are working on a fix here already: [NebraLtd/hm-diag#80](https://github.com/NebraLtd/hm-diag/issues/80)

Thank you for the fast reply, Kevin. Looking forward for the update. Cheers, have a good one.

closed 2021-08-05 by Vasilev87

SkyThommy, 2021-08-05:
@Vasilev87  Its not just a display error, because i wanted to run discovery mode, so i know where is the best place to assert my HS, but i cant do that because of this