NebraLtd/Helium-Guides #133: On onboard key found.

Pripadnal, 2021-08-06:
Dear Customer Support,

I have a specific issue related to 'no onboard key found' during the setup. it was not solve after the troubleshoot.

All the necessary data is provided via email and have opened ticked #259647. So far i dont have any solve of the issue.

The Nebra is sitting on the desk and waiting.

Could you please assist for faster solve.

Best Regards,
Miro

salmanfarisvp, 2021-08-11:
@Pripadnal We just added the onboarding key. can you please try to setup and see . please let me know if you are still facing any errors.  You are setting the device first, please take a look at "[The First Few Days – What To Expect](https://helium.nebra.com/handy-guides/first-days)". Thanks

closed 2021-08-11 by salmanfarisvp

SkyThommy, 2021-08-11:
@salmanfarisvp Can you please also check ticket #260727? Have exactly the same issue, thanks very much!

salmanfarisvp, 2021-08-13:
@Roomer562 Thanks for sharing the details over mail, please keep it connect with the internet we will be performing a fix soon and I'll let you know once it is completed over mail.