NebraLtd/Helium-Guides #136: Nebra dashboard - can't add device

dakky221, 2021-08-09:
I tried all possible combinations of browsers, manual input, QR code input, from phone and PC and laptop... it always says the device doesn't exists. Been trying for the last week every day at different times.

And then today it told me it soon won't be free anymore. Well, since it doesn't work anyway, I guess I'm not missing anything, but would like to see it before it starts to be paid.

Of course no one is responding on email. Ticket id 261949

salmanfarisvp, 2021-08-11:
Hi @dakky221, Sorry to hear about this issue. We just responded to your ticket, please check and provide the mentioned details, we will try to solve the issue ASAP.

closed 2021-08-11 by salmanfarisvp

dakky221, 2021-08-11:
It's Nebra Indoor. EU868. RPI serial and MAC you have in first email.