NebraLtd/Helium-Guides #139: Nebra: Miner Still Loading Sync/Height error

Spartan268, 2021-08-11:
Hi there,

I’m having trouble getting synced up to the network. I’ve run diagnostics on the miner and found that sync percentage and height percentage is listed as “Miner is Still Loading” Miner connected to blockchain is also listed as no. I’ve tried restarting the miner by plugging it out for a little while but this has not worked. Can you please help me out? 

Thanks,
Spartan

facilitator999, 2021-08-11:
I'm going crazy, mine is the same, fixes then hour later miner is loading.

Dashboard says synced and connected to block

salmanfarisvp, 2021-08-11:
@Spartan268 @facilitator999  Please share the below-mentioned details to sales@nebra.com, if already shared please share the ticket number here, will check. 

```
Model of unit (Indoor or Outdoor)
Mac address of the unit (Shown as ETH on sticker)
Frequency of the unit (Shown as Freq on sticker)
What country are you using the miner in?
How are you connecting it to internet? (Ethernet, Wi-Fi, Cellular)
How are you powering the unit? (Included adaptor, POE, Third party adaptor)
```

closed 2021-08-11 by salmanfarisvp

Spartan268, 2021-08-11:
262771

facilitator999, 2021-08-12:
#262712

salmanfarisvp, 2021-08-13:
@Spartan268 @facilitator999 I need more information about your device, please check the mail and please share the information mentioned.

Spartan268, 2021-08-13:
> @Spartan268 @facilitator999 I need more information about your device, please check the mail and please share the information mentioned.

I have messaged you over everything in our earlier chat on Discord.

facilitator999, 2021-08-13:
I'm banned from your discord for no good reason!

Spartan268, 2021-08-15:
Hi there,

I really need help with this. I’ve provided 
all the details. I cannot connect to the Helium App diagnostics despite all attempts so I’ve been pulling diagnostics from the miners IP. I’ve restarted, switched to Ethernet, disabled Bluetooth on my phone reconnected. I’ve had the miner for almost 7 days and it still hasn’t synced. The miner has been sitting currently for three days connected to Ethernet where I haven’t touched it in hopes it would sync. I was hopeful the issue would be solved as I was informed by Kevin in technical support that it should be up and running in a matter of hours but unfortunately not. Currently it’s sitting in a status of needs attention despite everything being connected and setup. I’ve noticed some days it connects to the network once and then won’t connect again at all that day. 

I really need this to be looked at and fixed as soon as possible. 

 ID #262614 Is my ticket number.