NebraLtd/Helium-Guides #142: Nebra Miner is Still Loading Issue

Spartan268, 2021-08-16:
Hi there,

I really need help with this. I’ve provided
all the details. I cannot connect to the Helium App diagnostics despite all attempts so I’ve been pulling diagnostics from the miners IP. I’ve restarted, switched to Ethernet, disabled Bluetooth on my phone reconnected. I’ve had the miner for almost 7 days and it still hasn’t synced. The miner was sitting for three days connected to Ethernet where it wasn’t touched in hopes it would sync. I was hopeful the issue would be solved as I was informed by Kevin in technical support that it should be up and running in a matter of hours but unfortunately not. Currently it’s saying it’s syncing. I’ve noticed some days it connects to the blockchain once and then won’t connect again at all that day. 

I really need this to be looked at and fixed as soon as possible.

ID #264141 Is my ticket number.

shawaj, 2021-08-16:
Please just leave it for a bit. And also ensure its not relayed.

Syncing takes time

Spartan268, 2021-08-16:
Absolutely Aaron, I agree and I’m happy to be patient when there’s a sound reason to be.. but it’s been 7 days I’m to believe syncing does not take that long. 

I have friends who received their miners after me and they are synced up and running already.

Looking at mine now on the App and it’s telling me  it hasn’t synced to the blockchain in 10 hours. It’s still showing as Miner is still loading in the diagnostics when I search the IP. 

Is there anyway you can remote into the miner or run an update or set it up from scratch so I can start the process all over again from the beginning? I really need a solution at this stage. 

Thank you for coming back to me.

shawaj, 2021-08-16:
@Spartan268 done a factory reset on your unit and it looks like it has recovered

It was suffering from this issue https://github.com/helium/miner/issues/978

Will keep an eye on it and hopefully it will sync fully now

Spartan268, 2021-08-16:
Thanks so much Aaron. Hopefully that does the trick. Can we leave the ticket open for the moment just in case?

shawaj, 2021-08-16:
For sure - i have left the ticket open and this issue and will keep an eye on it 👍

Spartan268, 2021-08-16:
Thanks Aaron. Just to give you a heads up.. it’s gone back to miner connected to blockchain false. I also tried running a diagnostic earlier on the app after your reset and it worked. Just tried to do it now and got the error again.

Kayz86, 2021-08-17:
My Nebra had been working just fine until yesterday afternoon it went into Blockchain false also.

closed 2021-08-17 by salmanfarisvp

shawaj, 2021-08-18:
@Spartan268 yours looks fine now. syncing well. However - your hotspot is relayed and this is probably why it is periodically losing connection to the blockchain. have a look here for how to fix this https://helium.nebra.com/handy-guides/port-forwarding/overview

@Kayz86 this is not necessarily an issue as long as it is synced

Spartan268, 2021-08-18:
> @Spartan268 yours looks fine now. syncing well. However - your hotspot is relayed and this is probably why it is periodically losing connection to the blockchain. have a look here for how to fix this https://helium.nebra.com/handy-guides/port-forwarding/overview
> 

Ok thank you for you help Aaron. At present I’m connected via Ethernet as it seemed to be dropping with wifi. Should I be able to go back to wifi once syncing is complete?

Kayz86, 2021-08-18:
> @Spartan268 yours looks fine now. syncing well. However - your hotspot is relayed and this is probably why it is periodically losing connection to the blockchain. have a look here for how to fix [...]

@shawaj Aaron I dont know what you mean it's not an issue? My Nebra is turned on and is showing on the dashboard and I can see the diagnostics page.

However because it's not on the blockchain and its showing 'FALSE' it has now stopped mining, appearing offline on the helium network. This was after your most recent firmware update.

It's now been offline from the blockchain for 2 days because its showing blockchain false. I have created a ticket also #265033.

Vasilev87, 2021-08-22:
Hi @shawaj hope you are well. I had my miner working fine for 2 weeks or so and since yesterday it is offline, not syncing and in the diagnosting showing that Miner Connected To Blockchain is False. When tried a few times to turn it off and on, yet nothing has changed and also when I try to run diagnostig through the Helium app I have this error: 

[image]

Had a chat with Spartan on discord and he told me it will require someone from the support team to look at the issue as it isn't something that can be user fixed.

Please advice where I should provide additional information. Here is the ticket number from the Nebra support: #267427