NebraLtd/Helium-Guides #145: Nebra firmware update has knocked my miner out

Kayz86, 2021-08-17:
I notice there is a new diagnostic page and with that a new firmware update version of 2021.08.16.1

I have noticed since this afternoon my miner has stopped mining,

These are the listed issues:

Not connected to the blockchain: False
Height Status: 0 / 968601
Modem Detected: False
Your help would be much appreciated.

salmanfarisvp, 2021-08-17:
Hi @Kayz86 , Can you please share the hotspot MAC address, Frequency details to sales@nebra.com. if already done, then please share the ticket number. 

I'll be closing this issue since it's not related to documentation.

closed 2021-08-17 by salmanfarisvp

Kayz86, 2021-08-17:
> Hi @Kayz86 , Can you please share the hotspot MAC address, Frequency details to [sales@nebra.com](mailto:sales@nebra.com). if already done, then please share the ticket number.
> 
> I'll be closing this issue since it's not related to documentation.

Hi Salman

Which mac address do you need there's two?

ETH0 MAC or WLAN0 MAC

Kayz86, 2021-08-17:
I have now sent you both, please see ticket: 265033

Thank you